pesor/TTGO-T-HIGrow #3: TTGO HiGrow with BME280

leeandy1, 2021-01-11:
I have just picked up 2 HiGrow boards and both have BME280 sensors onboard which are reporting null values. Both a version 1.1 boards 19-8-22.

How do I get these sensors to report values?

pesor, 2021-01-11:
There are no BME280 sensors on the boards, in the original program from LILYGO they referenced the BME280, but it is not there. 

The code now have taken that into account, and therefore it will not (and cannot) report data from a nonexisting BME280.

leeandy1, 2021-01-11:
[image]

Thanks for the response. This is a picture of the board I have. As you can see there is no DHT11 sensor. I understand the BME280 sensor is circled in red? Please do correct me if I am wrong.

leeandy1, 2021-01-12:
Just to confirm firmware V4.0.6 appears to all function correctly for all other sensors. I currently have the board running to test battery life and will report back. I purchased the full kit with housing and battery. I am happy to assist with any testing you require but would advise I am fairly new to this.

pesor, 2021-01-12:
I will look into this. 

The DHT sensor is discontinued, so I will order one with this sensor, and correct the coding to include BME280 also.

pesor, 2021-01-12:
I have just ordered on module from AliExpress, will amend coding when received.

leeandy1, 2021-01-12:
That's great to hear. If you need any help testing give me a shout.

pesor, 2021-01-12:
I also have ordered the BME280 chip it self, in order to try to solder it onto the board. 

There are som confusing in the naming, some Call it BEM280, others Call it BME280, anyone knows why?

pesor, 2021-01-12:
@leeandy1 thanks for confirming, that the boards works with the soil and salt sensor.

leeandy1, 2021-01-12:
@pesor I believe it is a BME280, not sure why some people call it a BEM280 - maybe a typo? 

My board has currently been running for 16 on a 200mAh battery and is reporting 97% left using your sleep settings. I need to take a look at boot count and sleep count as I think these are mixed up somehow. Boot count is 16 but sleep count is 0. Does your sleep routine effectively make the board boot each time it comes out of sleep?

pesor, 2021-01-12:
I have changed my interval to 3 hours om order to prolong the battery life. 

Boot count is everytime the module starts normally after a normal deepSleep. 

Sleep count, is when the module takes a 5 minutes DeepSleep, when something has gone wrong in the run. Like eg no network or mqtt connection. Zero is good, it should always be zero, otherwise the module do not run properly. The battery date counter does not work properly, I am working on that.

elecoq129, 2021-01-12:
> 
> 
> I will look into this.
> 
> The DHT sensor is discontinued, so I will order one with BEM280, and correct the coding to include BEM280 also.

On my side, I have order a pair DHT22 (white plastic packaging). I will do the soldering as soon as it arrives in replacement of the very cheap DHT11

pesor, 2021-01-12:
The DHT22 should work, if not, I will solder one (I have Them) to a module and check coding.

elecoq129, 2021-01-12:
> 
> 
> I have changed my interval to 3 hours DeepSleep to prolong the battery life.
> 
> Boot count is everytime the module starts normally after a normal deepSleep.
> 
> Sleep count, is when the module takes a 5 minutes DeepSleep, when something has gone wrong in the run. Like eg no network or mqtt connection. Zero is good, it should always be zero, otherwise the module do not run properly.
> 
> The battery date counter does not work properly, I am working on that.
> 
> When the module goes out of sleep, it follows the EPS32 standard, I just names it boot count by choise/coincidence.
> 
> When waking up from sleep, it fetches some variables stored, ad in a real boot these variables vil always be zero.

I'm using the BootCount value and it works. However, the Sleep-Counter doesn't? But I'm still using an early firmware release.

elecoq129, 2021-01-12:
however I am still desperate to get the external  PT100 sensor using the DS18B20, but still missing the correct value template for mqtt broker. I have tried {{ value_json.DS18B20.Temperature }} but receive "unknow values" while all other values works. Could you please help?

pesor, 2021-01-12:
I am my self not totally sure if the sleep5counter works, as I have not yet seen a count > 0, i implemented it because I at a certain point had an unexplainable battery drain.

elecoq129, 2021-01-12:
> 
> 
> I am my self not totally sure if the sleep5counter works, as I have not yet seen a count > 0, i implemented it because I at a certain point had an unexplainable battery drain.

To be honest, not really needed, as bootCount is much enough

pesor, 2021-01-12:
> however I am still desperate to get the external PT100 sensor using the DS18B20, but still missing the correct value template for mqtt broker. I have tried {{ value_json.DS18B20.Temperature }} but receive "unknow values" while all other values works. Could you please help?

I do not know what you are referring to ?

leeandy1, 2021-01-12:
You can add an external sensor to the I2C UART port. I think the PT100 is just an external waterproof DS18b20 temperature and humidity sensor.

elecoq129, 2021-01-12:
> I do not know what you are referring to ?

The HiGrow board has en external (white) connector just below the USB-C, which you can connect external devices. One of them is a waterproof PT100 sensor. When I used first the code from https://github.com/Xinyuan-LilyGO/TTGO-HiGrow it worked, and used the DS18B20 in the code. That part i currently missing in yours.

pesor, 2021-01-12:
@elecoq129 

please move this issue about the PT100 to a new issue, and I will look into it.

elecoq129, 2021-01-12:
> 
> 
> @elecoq129
> 
> please move this issue about the PT100 to a new issue, and I will look into it.

Ok will do. More information about my project, I am growing mushroom (in France) and need crucially soil hygro information (very important to me) as well as soil temperature. Reason why I desperately need this PT100 to work as well.

elecoq129, 2021-01-12:
Last but not least, I saw your new code release v.4.0.6, very difficult to integrate it into my current code (earlier firmware version). Would it be possible for you to generate a full .ino file to be able to start a new "Arduino" install from scratch?

elecoq129, 2021-01-12:
> 
> 
> You can add an external sensor to the I2C UART port. I think the PT100 is just an external waterproof DS18b20 temperature and humidity sensor. https://www.aliexpress.com/item/4000402593081.html?gps-id=pcStoreJustForYou&scm=1007.23125.137358.0&scm_id=1007.23125.137358.0&scm-url=1007.23125.137358.0&pvid=4fb3238d-4436-414b-b158-76a9fd96801b&spm=a2g0o.store_home.smartJustForYou_237836912.7
> 
> Adding support for external DS18b20 sensors would be really useful for measuring the temperature of the soil rather than the air temperature that the integrated sensor measures.

> I2C UART port

Yes correct this is the device (3 different variants)

pesor, 2021-01-12:
> Last but not least, I saw your new code release v.4.0.6, very difficult to integrate it into my current code (earlier firmware version). Would it be possible for you to generate a full .ino file to be able to start a new "Arduino" install from scratch?

You will need to install VSCode and PlatformIO, as described in the Readme. 

The Arduino EDI, I am no longer going to use, as the VSCode EDI, makes development and maintanance much more easy. 

It is not difficult to implement, there some very good YouTube videos, that shows how to do it. Just search for VSCode and then PlatformIO, and the will popup.

elecoq129, 2021-01-12:
> You will need to install VSCode and PlatformIO, as described in the Readme.
>
> The Arduino EDI, I am no longer going to use, as the VSCode EDI, makes development and maintanance much more easy.
>
> It is [...]

Ok I will look at it, just some more work...
Thanks for your support

pesor, 2021-01-12:
Let me know if you run into problems and I will try to help.

leeandy1, 2021-01-13:
> Ok I will look at it, just some more work...
> Thanks for your support

Don't be scared by VSCode and PlatformIO, i'm a complete newbie to this and managed to get up and running in no time.

elecoq129, 2021-01-13:
Ok Guys you won...thanks for the encouragement. I'll do it and let you know

elecoq129, 2021-01-14:
> 
> 
> Let me know if you run into problems and I will try to help.

Hi Pesor, I managed to install the new environment. No problem detected during compiling, flashing is ok. But here's some error error messages from the serial console?

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Void Setup
Listing directory: /
  FILE: /dash/favicon.ico	SIZE: 13532
  FILE: /dash/index.html	SIZE: 482
  FILE: /dash/js/app.js.gz	SIZE: 198509
- failed to open file for reading
Here comes the calibration info:

- failed to open file for reading
Here comes the name info:

- failed to open file for reading
Here comes the calibration info:

last cnt date 
Size of SSID array 2
[E][WiFiSTA.cpp:124] begin(): SSID too long or missing!

SSID name: Livebox_A5FE.............................................................
............................................................
............................................................
....................we should break
SSID name: KuWfi Back.. 
Connected to network
7C:9E:BD:FB:F2:C8
192.168.6.145

Any idea?

pesor, 2021-01-15:
Hej @elecoq129 

See my comments below to your Serial messages:

ets Jun 8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
**Until here - just some information I think, never bothered my self**

Void Setup
Listing directory: /
FILE: /dash/favicon.ico SIZE: 13532
FILE: /dash/index.html SIZE: 482
FILE: /dash/js/app.js.gz SIZE: 198509
**This is a listing of the files you have on the modules file system**

failed to open file for reading
Here comes the calibration info:

failed to open file for reading
Here comes the name info:

failed to open file for reading
Here comes the calibration info:
**These messages, is because you have not initialized the module, you will have to enter the correct information, according to the info in the file - user-defined-variables.h - which you find in the include folder, when done that, the files will be read, and content show.**

last cnt date
**Just info - not updated because of the above**

Size of SSID array 2
**Number of SSID's**

[E][WiFiSTA.cpp:124] begin(): SSID too long or missing!
**Ignore this, it has no meaning, for the program execution**

SSID name: Livebox_A5FE.............................................................
............................................................
............................................................
....................we should break
**Change to next SSID, because the first did not connect**
SSID name: KuWfi Back..
Connected to network
7C:9E:BD:FB:F2:C8
**MAC id of the module**
192.168.6.145
**IP of the module**

I hope this clarify for you, otherwise, do not hesitate to ask again.

Per

elecoq129, 2021-01-15:
> Hej @elecoq129
>
> See my comments below to your Serial messages:
>
> ets Jun 8 2016 00:22:57
>
> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee [...]

thanks for your support. Since then, I have sorted out my WiFi connexion issues. Quick one, does your soft support name resolution? eg rather than giving a IP address for the mqtt broker, can you use an name instead?

pesor, 2021-01-15:
Yeh I Think you can give a name in stead, try it and let me know. 

The MQTT connection, is a library, and not done by me.

elecoq129, 2021-01-17:
I'm still struggling with the new (latest firmware 4.0.6) config: can't get a connexion to the mqtt broker:

last cnt date
Size of SSID array 1
[E][WiFiSTA.cpp:124] begin(): SSID too long or missing!

SSID name: freebox_noyarey... 
Connected to network
7C:9E:BD:FB:F2:38
192.168.1.153
[E][WiFiUdp.cpp:170] beginPacket(): could not get host from dns: 2
[E][WiFiUdp.cpp:170] beginPacket(): could not get host from dns: 11
[E][WiFiUdp.cpp:170] beginPacket(): could not get host from dns: 11

So you can see I'm correctly connected to the WiFi network but not the mqtt server?
With the same parameter with the old firmware, everything is fine: meaning my server is up-and-running?
Any help from someone...

pesor, 2021-01-17:
@elecoq129 
It is not the MQTT broker that is the issue, the error is coming here:

WiFiUDP ntpUDP;
NTPClient timeClient(ntpUDP);

This is line 89 and 90 in main.cpp..

This function contacts the Time Syncronation Server over the Internet, and the message is strange, as it refers to some
DNS servers, which I cannot recognize, A dns server is an IP, that is like this nnn.nnn.nnn.nnn.

The error messages DNS, can though be part of the programs printed Error message, and thus referring to something in WiFiUdp.cpp.
(don't look into this program, it is a standard C++ program, that I have not made og changed in)

You definitively have an Internet connection error, as I see it, you are probably not connected to the Internet, or your WIFI is referring to wrong DNS's, options are multible, but have nothing to do with the program for the module (I think and expect)

So log on to the WIFI on your Smart Phone, use the same credentials, and see if you can connect to the Internet.

If that works, you have to check your settings in the User-settings .h file.

Don't hesitate to return, if you cannot solve it.

elecoq129, 2021-01-17:
> @elecoq129
> It is not the MQTT broker that is the issue, the error is coming here:
>
> WiFiUDP ntpUDP;
> NTPClient timeClient(ntpUDP);
>
> This is line 89 and 90 in main.cpp..
>
> This function contacts the [...]

Thans for your ananlyses. But what it's strange though, is with exactly (user variables) using the same configuration, my internet connections work well with the old version of the firmware?

pesor, 2021-01-17:
Try to comment the two Lines out in main.cpp, and restart. 

You will get errors, but they will be later in the program   

I help you through.

pesor, 2021-01-17:
Tou might have to comment other Lines out in order to compile, do that. When the program compiles run it and send me the error output.

elecoq129, 2021-01-17:
> 
> 
> Try to comment the two Lines out in main.cpp, and restart.
> 
> You will get errors, but they will be later in the program
> 
> I help you through.

I ave tried something else since your last recommendation. As I have 2 ISP, I changed the network configuration to connect to the other one (Livebox-A5FE vs. freebox_noyarey), guess what, it works!

SSID name: Livebox-A5FE....................... 
Connected to network
7C:9E:BD:FB:F2:C8
192.168.1.26
IN VINTERTIME
Wire ok
BH1750 Advanced begin
lux 0.00
0
Soil before map: 3399
needed
Volt direct 2701
Battery Voltage: 4.79
Writing file: /batinfo.conf
- file written
lux 63.33
Creating JSON document...
chipId 7c9ebdfbf2c8
Tgrow_HIGrow/7c9ebdfbf2c8
Livebox-A5FE
Sending message to topic: {"plant":{"Tgrow_HIGrow":"7c9ebdfbf2c8","sensorname":"","date":"17-01","time":"11:13","sleep5Count":0,"bootCount":0,"lux":63.33333,"temp":23.1,"humid":43,"soil":65535,"salt":0,"saltadvice":"needed","bat":100,"batcharge":"charging","batchargeDate":"17-01","batchargeDateCnt":0,"battvolt":2701,"battvoltage":4.788586,"wifissid":"Livebox-A5FE","rel":"4.0.6"}}
Attempting to connect to the MQTT broker: 192.168.1.47
You're connected to the MQTT broker!

pesor, 2021-01-17:
Good, so it was the DNS issue 🙂

elecoq129, 2021-01-17:
so you were right, it's definitely a router configuration issue, I will work this out later. 
My next question is I did not fully understand how to implement the new "autodiscovery" part:

 therefore run this program in Windows 10, and I do it in the Eric API for Python. The good thing is that you only need to run this program, when you add new modules. When it has run, it have made the mqtt messages from the TTGO-T-HIGrow module, able to be Autodiscovered by Home-Assistant MQTT Autodiscover function.

Do you have a tuto for it?

elecoq129, 2021-01-17:
> 
> 
> Good, so it was the DNS issue 🙂

yes indeed, for what ever in my DHCP router, for whatever reason it was referring to a (non-existing) IP address.
However I tried to replace the IP address value for the broker information it does not work. 
I have tried few syntax, no way, eg:

const char broker[] = "myhttpservice.fr/";
int        port     = 8123;

or 

const char broker[] = "http://myhttpservice.fr/";
int        port     = 8123;

could you please have a look at it, or do you want me to open a new issue?

pesor, 2021-01-17:
Your port no. is wrong

normally it is 1883 or 8883

Try that.

elecoq129, 2021-01-17:
> 
> 
> Your port no. is wrong
> 
> normally it is 1883 or 8883
> 
> Try that.

I tried both ports: 1883 & 4 no chance!

elecoq129, 2021-01-17:
This is what I get:

Here comes the calibration info:
17-01:0:17-01
last cnt date 17-01
Size of SSID array 1
[E][WiFiSTA.cpp:124] begin(): SSID too long or missing!

SSID name: Livebox-A5FE.. 
Connected to network
7C:9E:BD:FB:F2:C8
192.168.1.26
IN VINTERTIME
Wire ok
BH1750 Advanced begin
lux 0.00
0
Soil before map: 3403
needed
Volt direct 2693
Battery Voltage: 4.77
Writing file: /batinfo.conf
- file written
lux 45.83
Creating JSON document...
chipId 7c9ebdfbf2c8
Tgrow_HIGrow/7c9ebdfbf2c8
Livebox-A5FE
Sending message to topic: {"plant":{"Tgrow_HIGrow":"7c9ebdfbf2c8","sensorname":"","date":"17-01","time":"20:16","sleep5Count":0,"bootCount":0,"lux":45.83333,"temp":24.1,"humid":44,"soil":65535,"salt":0,"saltadvice":"needed","bat":100,"batcharge":"charging","batchargeDate":"17-01","batchargeDateCnt":0,"battvolt":2693,"battvoltage":4.774403,"wifissid":"Livebox-A5FE","rel":"4.0.6"}}
Attempting to connect to the MQTT broker: http://homeassistant.morillesdefrance.fr
[E][WiFiGeneric.cpp:654] hostByName(): DNS Failed for http://homeassistant.morillesdefrance.fr
-2
Going to sleep... 300 sekunder

pesor, 2021-01-17:
That is the problem,

http://homeassistant.morillesdefrance.fr

it should probably be http://morillesdefrance.fr

I do not know how it gets the homeassistant. first, unless your broker is running in a subdomain under morillesdefrance.fr.

You should check that.

elecoq129, 2021-01-17:
> http://morillesdefrance.fr
Whatever I'm trying:

morillesdefrance.fr
or
/morillesdefrance.fr
or
http://morillesdefrance.fr

below the error message

Attempting to connect to the MQTT broker: http://morillesdefrance.fr
[E][WiFiGeneric.cpp:654] hostByName(): DNS Failed for /morillesdefrance.fr
MQTT connection failed! Error code = -2
Going to sleep... 300 sekunder

pesor, 2021-01-18:
@elecoq129 

Now I know what is wrong.

I tried to ping morillesdefrance.fr, and that works, I get response so no problem with the URL.

Then I took a closer look at your message when you used the IP number, and here is the explanation.

You have your mqtt broker on your LAN, which is OK, but you do not connect to it, when trying via the URL, and there is an explanation for that.

You must make a **port forward in your router** on port 1883, that refers to IP 192.168.1.47, which is your IP of your MQTT broker on your LAN.

If you do this, you open your MQTT broker for EVERYONE on the Internet, so you have to be very sure about USERID and PASSWORD, as your LAN otherwise is WIDE OPEN to the LAN.

I do not understand, why you want to use the URL, it is much safer just to use your LAN ip, as no information then is sent outside your LAN to the INTERNET (WORLD), but kept locked inside your safe LAN.

I have my MQTT broker on my Synology, and I would never address it via my URL's.

Per

elecoq129, 2021-01-18:
> @elecoq129
>
> Now I know what is wrong.
>
> I tried to ping morillesdefrance.fr, and that works, I get response so no problem with the URL.
>
> Then I took a closer look at your message when you used the IP [...]

Hi Per, thanks again for you deep analyses. First of all, the reason why I want/must use an URL vs. an straight IP address is very simple: my green houses are not in the same LAN. All of them are equipped with a 4G/5G WiFi router. Thus I need to go through routing. 

Of course I did set-up a port forwarding in my ISP router (I'm using the one that provides a IP fix, thus no need of using DDNS), where the mqtt server is located. And yes I have special users defined with no supervisor rights for both mqtt and HomeAssitant. In term of  configuration, I have a WIN10 PC with Ubuntu VM (Oracle) + HA as well as Synology server. For now I'm using my PC, which works well, as I'm facing issues with the hassiao & mqtt configuration with Synology. In a longer term, plan is to replace my current desktop by a mini PC dedicated (10w low consumption => https://www.aliexpress.com/item/1005001650733496.html?spm=a2g0o.cart.0.0.33373c00Yr2Dpp&mp=1) 
Security wise, in a longer term, I will look at implementing some security : SSL. I'm not sure the current firmware as a module for it?
I will check later on why I can't get the mqtt connection using a URL. As said it's looks like a detail, it's very important to me.

pesor, 2021-01-18:
I understand why you need the external access. And I see that you have Security in place. 

Have you tried to use the external IP in stead of the URL, not as a permanent solution, unless you as I have, Got a fixed IP. 

If it works with the external IP, your problem is a DNS issue, if it does not work, I still Think that something is wrong with your port forwarding. 

Do you have a firewall on the router, it can be that blocking the communication. 

I understand that the router you use is the ISP’s. I NEVER do that, I only use the ISP’s router as a MODEM, and I have a Synology router to handle my Traffic. 

My experience is that the ISP’s routers NEVER works optimal. 

I can recommend Synology or Asus they are very good.

pesor, 2021-01-18:
Also the Security on your PC can be the reason for blocking, if your port 1883 is not open to the internet, you will not get the communication.

elecoq129, 2021-01-18:
> Also the Security on your PC, it that is running your mqtt can be the reason for blocking, if your port 1883 is not open to the internet, you will not get the communication.

Ok, I finally sorted out the issue. In my local ISP router, I was only redirecting tcp port 8123 in order to access HommeAssitant (VM). I just added another tcp port redirection 1883 and now it works. So I confirm the standard WiFi library does name resolving.

pesor, 2021-01-18:
@elecoq129 
As this is not really connected to the BME enhancement I will delete our conversation. 

Is that OK with you?

elecoq129, 2021-01-18:
> 
> 
> @elecoq129
> As this is not really connected to the BME enhancement I will delete our conversation.
> 
> Is that OK with you?

Absolutely agree: network configuration issue. Thanks again for your time & effort.
Did you receive you device for testing the DS18B20 UART port?

pesor, 2021-01-18:
The BME280 and temp sensor er on it’s way from China

elecoq129, 2021-01-18:
> The BME280 and temp sensor er on it’s way from China

one more question now on the autodiscovery configuration. When you say:
T_he downloaded folder Autodiscovery, you copy to Home Assistant config folder (where you have your configuration.yaml)__

Do you mean copy the contents of the autodiscovery into config from HomeAssistant or you have to create also the directory structure and the files?

pesor, 2021-01-18:
@elecoq129 

Actually, the placement of the autodiscover folder, can be of your own choice.

You need to have the sensors folder, and the sensor.yaml in that folder.

In the sensor.yaml, you erase all but the first 4 lines, and a blank 5th line.

You run the TTGO-T-HIGrow-aut.py, as you normally would run a Python script.

I use the EDI - Eric6, that is a fantastic Python EDI.

elecoq129, 2021-01-18:
> EDI

I assume that You have to run the aut.py script in the win10 machines that runs HomeAssistant?
If I manyuall chage the mac information in the sensor.yaml, that would not be enough for HomeAssistant to get the information from the sensor?

pesor, 2021-01-18:
@elecoq129 

I have moved your last question to a new Issue:

https://github.com/pesor/TTGO-T-HIGrow/issues/6#issue-788284260

I am happy to answer any questions, but I prefer that a new issue is created whenever there is a question about functionality of the module.

leeandy1, 2021-01-18:
@pesor just a bit of an update on battery usage. Currently the device has been running for 146 hours and I still have 62% battery left. I'm using the 200mAh battery that's included in the kit from LilyGo.

ericsebastion, 2021-01-19:
I just purchased the BME280 version and i cant find where to download the arduino file for this new 4.x version I was redirected to. I used the old version and the soil part shows up as working in the serial monitor but the mqtt wont connect and I don't see the temp and humid readings. Any help would be appreciated. I'm trying to use this in Home Assistant also if that helps clarify.

[4 more replies]

closed 2021-03-03 by pesor