pesor/TTGO-T-HIGrow #5: external PT100 sensor

elecoq129, 2021-01-12:
the idea is to make use of of the external I2C UART port, adding support for DS18b20 sensors would be really useful for measuring the temperature of the soil rather than the air temperature that the integrated sensor measures.
As current firmware does not use, could you please implement it?

Thanks & Regards

pesor, 2021-01-12:
You can add an external sensor to the I2C UART port. I think the PT100 is just an external waterproof DS18b20 temperature and humidity sensor. https://www.aliexpress.com/item/4000402593081.html?gps-id=pcStoreJustForYou&scm=1007.23125.137358.0&scm_id=1007.23125.137358.0&scm-url=1007.23125.137358.0&pvid=4fb3238d-4436-414b-b158-76a9fd96801b&spm=a2g0o.store_home.smartJustForYou_237836912.7

Adding support for external DS18b20 sensors would be really useful for measuring the temperature of the soil rather than the air temperature that the integrated sensor measures.
I2C UART port

Added for my info.

pesor, 2021-01-13:
I have ordered the item, and will implement it when I have received Them

elecoq129, 2021-01-14:
Hi Pedro I'm planing to work tonight. Any URL to advise me for a good tuto on how to use PlatformIO IDE?

closed 2021-03-03 by pesor