pesor/TTGO-T-HIGrow #7: 4.0.6 - Arduino --> VSCode and PlatformIO

pesor, 2021-01-19:


pesor, 2021-01-19:
I did abandon the Arduino EDI for practical reasons.

I have changed to VSCode and PlatformIO, as this platform is much easier to maintain, and do development in.

You will have to install VSCode and PlatformIO, in order to run version 4.x.x.

On YouTube there is very good instructions on how to do this, just search for VSCode and PlatformIO, and it should go very easy to install it, and after it is installed, it is like using any text editor, this is just preperet for C++, and it is very easy to compile and get the programs to the module.

Don't hesitate to ask questions in this thread, I will answer fast.

pesor, 2021-01-19:
@BeardedTinker 

Maybe it is soon time for a video for the 4.x.x installation, with an inclusion for the VSCode and PlatformIO

;-)

pesor, 2021-01-19:
I came by this video, it is about 3D firmware update, but there are a chapter where he explains how to install VSCode, and another chapter for the PlatformIO.

All the rest you just skip.

https://www.youtube.com/watch?v=eq_ygvHF29I

BeardedTinker, 2021-01-19:
> @BeardedTinker
> 
> Maybe it is soon time for a video for the 4.x.x installation, with an inclusion for the VSCode and PlatformIO

It's overdue already! I hope, if all goes as planed, to have it done in first week of February. Kids should be back in school then (real not online) and I'll have a bit more time then. 
I'm really guilty for this one 😁

ericsebastion, 2021-01-19:
@BeardedTinker looking forward to it!

elecoq129, 2021-01-20:
If I may a comment to BeardeTinker, as there is a large of o tuto for installing VSCode/PlattformIO (https://randomnerdtutorials.com/vs-code-platformio-ide-esp32-esp8266-arduino/) I would rather prefer you'll work on a tuto to install the necessary stuff (Eric + Python) in order to run the Autodiscovery part. That would me usuful!
It's just an idea...

closed 2021-03-03 by pesor