pesor/TTGO-T-HIGrow #8: 4.x.x integration HA on Raspberry PI

pesor, 2021-01-22:
@ericsebastion 

Lets start with the basics.

You are running Home Assistant on Raspberry PI!
Which version of PI?

Do you have a mqtt server running on the Raspberry PI?

Do you have a Windows10 PC, as a workstation?

This is 3 crucial questions - I need answer for, to proceed.

ericsebastion, 2021-01-24:
I have a Raspberry Pi 4+ and using Mosquito Broker for my MQTT. Im using a Windows 10 to do my code work and flashing of other ESP32s i have for my grow tent.

Thanks!

pesor, 2021-01-25:
This is what I expected.

You then only need to use Windows10 to do all the implementation about the TTGO-T-HIGrow.

You need to have VSCode and PlatformIO installed, there is a good tutorial video in the issue about VSCode and PlatformIO.

You need to install Python for Windows10, and Eric6, there is a documentation made by Emmanuel, in the Repo Code section.

Then you should be ready to start running.

Start with the modules, and use MQTTlens, to see the output from the modules on your mqtt server(broker)

You will not see any modules in HA, until autodiscover, but that is ALL as it should be.

When you have that up and running, and see the messages in MQTTlens, we go to the autodiscover part.

//Per

pesor, 2021-01-25:
@ericsebastion 

I just reread my above note, and it could be read as you are on your own on the first part.

This is not the case, when doing the first part, just ask as many questions as you need, I will be there.

//Per

closed 2021-03-03 by pesor