pesor/TTGO-T-HIGrow #9: Core panic

crsprsp, 2021-03-22:
My Board tells me:
---------------
SSID name: HelloG............ 
Connected to network
0C:9E:9D:9B:92:9C
10.10.17.6
IN VINTERTIME
Wire ok
Could not find a valid BMP280 sensor, check wiring!
BH1750 Advanced begin
lux 27.50
0
Soil before map: 3523
Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.
-----------------
And reboots.
Tried with another (new) Board but it's the same.

pesor, 2021-03-22:
Are you on 4.3.1

crsprsp, 2021-03-23:
Yes, i am. Also downloaded the V4.3.1 again and uploaded it again. Same behavior.

pesor, 2021-03-23:
You are getting a division by zero error, 
It is something in your user settings, I thing

Have you made the BME variable false?

I need your user-variables file, remember to remove your passwords.

check main.cpp for the following

const String rel = "4.3.1"; // Finally the days since last charging works correctly.

// mqtt constants
WiFiClient wifiClient;
PubSubClient mqttClient(wifiClient);

// Date calculator
unsigned long epochTime;
String battChargeEpoc;
unsigned long epochChargeTime;
float battChargeDateDivider = 86400;

The battChargeDateDivider must be 86400

crsprsp, 2021-03-23:
ok, here we go: 

// *******************************************************************************************************************************
// START userdefined data
// *******************************************************************************************************************************
#include <Arduino.h>
// Turn logging on/off - turn read logfile on/off, turn delete logfile on/off ---> default is false for all 3, otherwise it can cause battery drainage.
const bool  logging = false;
const bool  readLogfile = false;
const bool  deleteLogfile = false;
String readString; // do not change this variable

// Select DHT type on the module - supported are DHT11, DHT12, DHT22 -- Even if you do not have one, you have to uncomment one, as compilation else will fail.
#define DHT_TYPE DHT11
//#define DHT_TYPE DHT12
//#define DHT_TYPE DHT22

// Set to true if you have a DHT sensor on the board, and false if not
bool dht_found = true;

// Select if 18B20 soil temp sensor available, if available -->> set to true
const bool USE_18B20_TEMP_SENSOR = false;

// It is a really good thing to calibrate each unit for soil, first note the number when unit is on the table, the soil number is for zero humidity. Then place the unit up to the electronics into a glass of water, the number now is the 100% humidity.
// By doing this you will get the same readout for each unit. Replace the value below for the dry condition, and the 100% humid condition, and you are done.

// Soil defaults - change them to your calibration data
int soil_min = 1535;
int soil_max = 3300;
bool calibrate_soil = false;

// Salt/Fertilizer recommandation break points. You can change these according to your own calibration measurements.
int fertil_needed = 200;
int fertil_low = 201;
int fertil_opt = 251;
int fertil_high = 351;

// Give the sensor a plant name, change to true, upload sketch and then revert to false
const bool update_plant_name = true;
String plant_name = "Tomato";

//If using the Greenhouse automatic watering repo, then assign a waterValveNo to the plant. 
int plantValveNo = 1;

// define your SSID's, and remember to fill out variable ssidArrNo with the number of your SSID's
String ssidArr[] = {"ThisIsMyWireless" };
int ssidArrNo = 1;

const char* ssid = ""; // no need to fill in
const char* password = "My1Passwo4D";
const char* ntpServer = "pool.ntp.org";

// Off-sets for time, and summertime. each hour is 3.600 seconds.
const long  gmtOffset_sec = 3600;

// Device configuration and name setting
const String device_name = "Tgrow_HIGrow"; // Can be changed, but not necessary, as it will give no added value.

#define uS_TO_S_FACTOR 1000000ULL //Conversion factor for micro seconds to seconds
#define TIME_TO_SLEEP  10800       //Time ESP32 will go to sleep (in seconds)

const char broker[] = "my.broker.com";
int        port     = 1883;
const char mqttuser[] = "ThisUser"; //add eventual mqtt username
const char mqttpass[] = "ThisPassword1"; //add eventual mqtt password

crsprsp, 2021-03-23:
And to be complete a snipplet from the monitoring output of the vcode:

SSID name: ThisIsMyWireless. 
Connected to network
7C:9E:BD:FB:F1:90
192.168.10.54
IN VINTERTIME
Wire ok
Could not find a valid BMP280 sensor, check wiring!
BH1750 Advanced begin
lux 30.00
0
Soil before map: 3729
Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.
Core 1 register dump:
PC      : 0x40163751  PS      : 0x00060c30  A0      : 0x800d1a0c  A1      : 0x3ffcab80
A2      : 0xfffa4f5c  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000064
A6      : 0xffffff9c  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x3ffcab60
A10     : 0x00000002  A11     : 0x00000e91  A12     : 0x0000000a  A13     : 0x00000000
A14     : 0x0000011e  A15     : 0x3ffc2494  SAR     : 0x0000001c  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff

ELF file SHA256: 0000000000000000

Backtrace: 0x40163751:0x3ffcab80 0x400d1a09:0x3ffcaba0 0x400d3817:0x3ffcabc0 0x400daca6:0x3ffcace0 0x40091272:0x3ffcad00        

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
Void Setup
Listing directory: /
  FILE: /name.conf      SIZE: 6
- failed to open file for reading
Here comes the calibration info:

Writing file: /name.conf
- file written
- failed to open file for reading
Here comes the charge date info info:

Size of SSID array 1

SSID name: ThisIsMyWireless.............
Connected to network
7C:9E:BD:FB:F1:90
192.168.10.54
IN VINTERTIME
Wire ok
Could not find a valid BMP280 sensor, check wiring!
BH1750 Advanced begin
lux 30.00
0
Soil before map: 3728
Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.
Core 1 register dump:
PC      : 0x40163751  PS      : 0x00060c30  A0      : 0x800d1a0c  A1      : 0x3ffcab80
A2      : 0xfffa4fc0  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000064
A6      : 0xffffff9c  A7      : 0x3ffc8da0  A8      : 0x00000000  A9      : 0x3ffcab60
A10     : 0x00000002  A11     : 0x00000e90  A12     : 0x0000000a  A13     : 0x00000000
A14     : 0x00000020  A15     : 0x3ffc2494  SAR     : 0x0000001c  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff

ELF file SHA256: 0000000000000000

Backtrace: 0x40163751:0x3ffcab80 0x400d1a09:0x3ffcaba0 0x400d3817:0x3ffcabc0 0x400daca6:0x3ffcace0 0x40091272:0x3ffcad00        

Rebooting...
ets Jun  8 2016 00:22:57

pesor, 2021-03-23:
I just took a bright new board, and tried to flash the board.

I get exact the same error as you.

Somehow, the current users have used boards with earlier versions of the firmware on them, and then avoid the error.

I will look into it immediately.

crsprsp, 2021-03-23:
perfect! glad you could reproduce the error! If you need extended testing, please gimme a sign!

pesor, 2021-03-23:
So, I found the reason, and I found a bypass..

Set the **bool calibrate_soil = true;**

it then writes the file, and you can change to false again, and change again to true, when you do the actual calibration, and then remember to set to false, when calibration has been done.

When the file do not exists, it creates a **division by zero error** in the program, this is the reason.

I will make a proper fix later.

pesor, 2021-03-23:
Thanks, but I know what is wrong, so I just need the time to program the fix.

In the meentime use the worksround in order to use the firmware.

Sendt fra min iPhone