pesor/TTGO-T-HIGrow #10: Autodiscover in Raspi 4

front0708, 2021-04-07:
Hi pesor,

thank you fpr the whole wiki. 

I really like to bring it to work on a my raspiberry pi. I studied again and again the youtube vids and the Wiki. It all works fine for me. The sensor is connecting to the MQTT broker und publishes the message succesfully. And then I stop because I don't know how to execute the "TTGO-T-HIGrow-aut.py" via python.

- Via putty: There ist no python to start on raspi for starting the "TTGO-T-HIGrow-aut.py"

 - Via Add-On "Visual Studio Code" in Homeassistant: There is the wrong python-version and therefore no paho-mqtt. 
 
 - Via eric IDE I don't know how to start the "TTGO-T-HIGrow-aut.py" on raspi, Home Assistant from the WIN 10 Eric IDE

I really don't want to throw the towel after those hours of trying and i'm very interested in getting things done. Can you please tell me, how to do it in raspi or is it not possible. When it is the second point, can you tell me whre to find how to integrate the sensor manually in home assistant? 

Thank you so much. Up to this point, I am very disappointed. If you can help me, It woul be a pleasure for me.

front0708

ivanling, 2021-04-10:
Hi,
I have the exact same problem. I have put TTGO-T-HIGrow-aut.py inside /python_scripts and sensors.yaml inside /python_scripts/sensors.yaml. Now I see Python Scripts: ttgo-t-higrow-aut in developer-tools/service ,the service dev tool allows you to call any available service in Home Assistant. But after that my sensor.yaml remains unchaged, and if I see Home Assistant log I have:

2021-04-10 23:07:29 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [548366779392] Error handling message: Service python_script.ttgo-t-higrow-aut does not match format <domain>.<name> for dictionary value @ data['sequence'][0]['service']. Got 'python_script.ttgo-t-higrow-aut'

Any idea?

Thanks

pesor, 2021-05-15:
Sorry for not answering earlier, I am right now very busy with my Greenhouse project and have not seen your input. 

First of all, the program cannot run inside Home Assistant, as Home Assistant do not support the file reading of the sensors.yaml. 

It is a simple Python script, which should run in any Python environment. 

You just have to address same mqtt server as Home Assistant, and the sensors.yaml should be accessed from the Autodiscover.py. 

The Autodiscover.py is completely Independent of Home Assistant, it just follows the Home Assistants Autodiscover routines. 

So if you can run any Python script on your Raspberry, you just run Autodiscover.py same way, and it should work. 

Again, do not try to run it via Home Assistant, it will NEVER work, as some of the features are not supported by Home Assistants build in service protocol.

pesor, 2021-05-15:
Found this for installing Eric in Raspberry PI, maybe it will help. 

https://github.com/REDUNION/eric_ide_on_raspberry_pi_4

ivanling, 2021-05-15:
Hello, Thanks for your answers. I have postponed switching to Linux, my entire digital life was Windows, but now I wanted to take this opportunity to learn and develop for home assistant to also develop the domain of competencies in Linux-based OS.
I already have Lubuntu 18.04 running in a ASUS EEE 1000H, and have managed to made an installation of HA, in python virtual environment. As soon as I have some time I go try to make things work with your firmware and with help of 
BeardedTinker tutorial

Thanks a lot