pesor/TTGO-T-HIGrow #13: In code remarks You're wondering what "Salt" is

wagnerohz, 2021-05-16:
The sensor measures the conductivity of the earth between the two shiny silver buttons. This is an indication of the amount of dissolved salts that the plants need as food. These values can only be approximations. And they only make sense if the soil is moist. The limit values for "needed", for example, seem to me to be set too high. 
I measured once:
- Fresh rainwater has 0 (measured value: 30µS)
- Tap water has 78 (measured value: 588 µS)
- Water with liquid fertiliser according to the manufacturer's instructions: 371 (measured value: 4000 µS)
The values of the sensor are fairly linear.

This is not a criticism of your magnificent work, just a (perhaps) helpful addition.

pesor, 2021-05-16:
Yes the salt figure is a tricky part. 

The tap water here in Denmark is filled with calcium, at my House the DH is > 26, the salt measurement in clean tap water in a glass is approx. 200. 

It is possible in uservariables to set the salt levels, so that it comply with your needs. 

I actually did an (not released yet) addition to the salt issue, where the figure was divided with the soil level/100, and a treshold for when calculation should start, eg >60% soil. 

It is giving me a division be Zero error sometimes, so I will look into it, and add it to next release. 

I am currently working on a special HIGrow Lovelace card, but that is to be in a much later release. 

Also there is a bug in the measurement, as it is done with DC current and not AC current, this means that the measurement will detoriate over time, if you do not make sure that your watering flushes the sensor. 

The DC current organise the ION's so that they are placed in ION- to one side and ION+ to the other side, which makes the measurement going toward zero. 

//Per

wagnerohz, 2021-05-16:
Thank you very much for the detailed engagement with my reflections. Yes, we have 6-7° DH here in my place in northern Germany. Now I understand why the threshold values in your code are suggested so high. 
Thank you for the important hint with the "sorting" of the ions at the anodes. I had also noticed that the values dropped faster than my test plants could "drink". I will observe the flushing of the sensor.

I am very curious about your next releases.

//Christian

pesor, 2021-05-16:
Hi Christian.

I have been working with a better presentation of the most important data from the sensor.

As I wrote I am trying to make a Lovelace card, but I have at this moment an intermidiate solution you might like.

I did send this to Bearded Tinker, with whom I communicate a lot.

————————-

This is the latest on my TTGO-HIGrow sensor output for my plants.

I have put development of a better card on hold, as I have too many things on my hand at the moment, so I figured this is a good intermediate solution.

I have included a screen shot, and the code for the “Drivhusplanter”, which is the most comprehensive.

Here are the HIGrow info.

[cid:6C21789F-894D-4A77-A1F4-83F47C5CEE5D]

Here are the code:

type: entities
title: Drivhusplanter
entities:
  - entity: sensor.padron_1_battery
    name: Padron 1
    type: 'custom:secondaryinfo-entity-row'
    secondary_info: >-
      <b style='color:red'> Last updated: [[ sensor.padron_1_date ]] at [[
      sensor.padron_1_time ]] <br> On battety: [[ sensor.padron_1_daysonbattery
      ]] days <br> Last charged: [[ sensor.padron_1_batchargedate ]]</b>
  - entity: sensor.padron_1_soil
    name: ' '
    type: 'custom:secondaryinfo-entity-row'
    secondary_info: >-
      <b style='color:green'> Gødningstilstand: [[ sensor.padron_1_fertilizer ]]
      <br> Jordtemperatur: [[ sensor.padron_1_soiltemp ]] <br>
      Drivhustemperatur:  [[ sensor.padron_1_temperature ]] </b>
  - entity: sensor.padron_3_battery
    name: Padron 3
    type: 'custom:secondaryinfo-entity-row'
    secondary_info: >-
      <b style='color:red'> Last updated: [[ sensor.padron_3_date ]] at [[
      sensor.padron_3_time ]] <br> On battety: [[ sensor.padron_3_daysonbattery
      ]] days <br> Last charged: [[ sensor.padron_3_batchargedate ]]</b>
  - entity: sensor.padron_3_soil
    name: ' '
    type: 'custom:secondaryinfo-entity-row'
    secondary_info: >-
      <b style='color:green'> Gødningstilstand: [[ sensor.padron_3_fertilizer
      ]]</b>

You need to install card-tools.js, and secondaryinfo-entity-row via HACS, read more here: https://github.com/custom-cards/secondaryinfo-entity-row

————————-

Best regards,
Per

Sendt fra min iPhone