pesor/TTGO-T-HIGrow #16: Fix Broker Port in save-configuration.h

Willi66, 2021-05-25:
error in save-configuration.h in line 77
  mqttClient.setServer(broker, 1883);
must be
  mqttClient.setServer(broker, port);
port is fixed 1883 instead of variable "port" in user-variables.h
  my server works on 1884 so I find this

in go-to-deep-sleep.h in 
line 7
    writeFile(SPIFFS, "/error.log", "Going to sleep for 10800 seconds \n");
is fixed 10800 and not the variable "TIME_TO_SLEEP" in user-variables.h

line 32
   Serial.println(" sekunder");
must be
   Serial.println(" seconds");

pesor, 2021-05-28:
You are right, will be changed in next release, primo June.

pesor, 2021-05-29:
I have committet your change, so it is corrected. 

Thanks

ppetermann, 2021-06-02:
I guess you haven't pushed it yet?

pesor, 2021-06-03:
I am not so experienced in Github, so I have to find out how to push it. 

//Per