pesor/TTGO-T-HIGrow #20: Home Assistant sensor config MQTT not retained and eventually sensors all become 'unavailable'

gitothy, 2021-06-27:
I've been seeing odd behaviour that all entities for my HIGrow show as _unavailable_ in Home Assistant after a period of time. I found this odd as the device is running from the mains and TIME_TO_SLEEP is at the default of 3 hours (10800 seconds).

I've managed to work out that the config entry to MQTT under topic homeassistant/sensor/Tgrow_HIGrow_macaddress isn't retained and seems to disappear at "some point".  I can see that my sensor had crashed on the latest occurrence of this issue and had been offline for 2 weeks while I was away.  I've just power cycled it and can see a new message on the broker but the sensors show as unavailable in HA.  Expected behaviour here would be that when the new data is pushed to MQTT and you're back running again.  Last time this happened there was no two week gap, although I can't remember exactly what happened other than that I had to restart the broker a few times to fix an issue that it didn't have the right write permissions to save it's database and log.

My workaround for whenever this happens is to republish the config using **TTGO-T-HIGrow-aut.py** which restores it in HA.  Must be a better way to deal with this though?

pesor, 2021-07-03:
Which MQTT broker are you using, as I have not seen this problem my self.

gitothy, 2021-07-06:
Mosquito in a docker, latest version

pesor, 2021-07-06:
I have experienced this my self. 

Most times was in connection with broker Update or Home Assistant upgrade, but I have not seen it for the last 3 months.  

I do not know what causes it, but yes, delete in Home Assistant, delete in sensors.yaml, return the Python Autodiscover program, and as you describe it All works again. 

Actually the same have happened with the TASMOTA devices, where on had to follow same procedure to get the devices back in Home Assistant. Also I have not seen this for 3 months.