pesor/TTGO-T-HIGrow #21: MQTT Connection failure

JohnnyNegativity, 2021-07-06:
Hi, I have been following bearded tinkerer's instructions to get this setup with home assistant.

I have the mosquitto MQTT broker running in home assistant, working for multiple other devices. 

When I configure the user variables and flash the board, it successfully connects to wifi but cannot reach the MQTT broker, giving the error below:

Sending message to topic: {"plant":{"Tgrow_HIGrow":"c44f337fc6e1","sensorname":"","date":"06-07","time":"15:46","sleep5Count":0,"bootCount":0,"lux":835.8333,"temp":26.3,"humid":42,"soil":65535,"soilTemp":0,"salt":0,"saltadvice":"needed","bat":100,"batcharge":"charging","batchargeDate":"06-07","daysOnBattery":0,"battvolt":2752,"battvoltage":4.879004,"pressure":0,"plantValveNo":1,"wifissid":"21BTG_NoT","rel":"4.3.2"}}
Attempting to connect to the MQTT broker: 192.168.1.100
[E][WiFiClient.cpp:258] connect(): socket error on fd 55, errno: 104, "Connection reset by peer"
MQTT connection failed! Error code = -2
Going to sleep... 300 sekunder

Any help would be much appreciated. Thanks!

pesor, 2021-07-06:
First thing that comes to mind is the broker port no. 

I have a typo in the Connect issue, where 1883 is hard coded, even though the port is part of usersetting. 

So if you are not using 1883, but another number this could be the reason. 

Next is username and password, if you are using this check for typing error. Mixed case / uppercase / both. 

//
Per

JohnnyNegativity, 2021-07-06:
Thanks very much for the quick response. 

Indeed I am using a non-standard port and this has solved the issue. 

Much appreciated!

elecoq129, 2021-07-22:
Hi Johnny,
As I have never experienced any issue with my envirnoment, like you described above, I will be very intereted to know what is your config? I am using a dedicated PC running Win10 + Unbuntu server (Oracle VM v. 6.1)  + HA and MQTT server (latest version). I have roughly 10 TTGO sensors connected to it using WiFi (WAN) connection. All is well.