pesor/TTGO-T-HIGrow #22: [Casing] Size-up hole + srew

FutureCow, 2021-07-09:
Really nice project, already using this to monitor some plants. 
Also want to monitor some outdoor plants, so was looking at your caseing, nice!

But isn't it easier to make the hole on the side a bit bigger, so you can just connect the USB-C on the port?

pesor, 2021-07-09:
Actually, I have changed so that the Hole fits to a C-plug, will upload it latest sunday.

FutureCow, 2021-07-09:
Thanks for the quick reply, looking forward to it!

pesor, 2021-07-11:
Probably first monday or tuesday

FutureCow, 2021-07-14:
Case ready for sharing?

pesor, 2021-07-15:
As I am very busy at this moment, I have uploaded the casing workfiles to [casing ](https://tordenvejr.nu/casing.zip),

This is my latest workfiles, and you will find the C-plug cacing here.

// Per

pesor, 2021-07-21:
@FutureCow

Have you downloaded

FutureCow, 2021-08-25:
Yes, sorry for the late reply. 
I can use the 3D printer next week, But wich is the file that you changed? 
Cattleya.stl
ClosedBatteryCase.stl
Padron2.stl
Padron3.stl