pesor/TTGO-T-HIGrow #23: waterproof casing

garudaonekh, 2021-08-24:
hi,
what is the recommended method to make it usable outdoor?

pesor, 2021-08-24:
I do two things.

First I spray the modules with and isolation and protection enamel/polish, in order to avoid corrotion.

I use the Teslanol for this.

Second I seal the case where the module sticks out, I use silicone, which is made for aquariums, and then I put some on where the plug is on the casing, as it can make a small opening. I use a finger with a little silicone, and smear it on.

It is easy to take apart again, if needed, for a new battery etc.

garudaonekh, 2021-08-24:
I think a 1" pvc pipe is the cheapest solution but the problem is the bottom part. it is difficult to protect the part that stick out

garudaonekh, 2021-08-24:
[image]

pesor, 2021-08-24:
Look at a previous issue, there is a link to my 3D print of cases.