pesor/TTGO-T-HIGrow #27: Auto configuration script error

riddlecp, 2021-09-30:
Hi, running into an issue with the auto discover python script.  Have tried multiple python versions just to be sure.  Any thoughts?

`python3 TTGO-T-HiGrow-aut.py
Connected with result code 0
in on_message
Tgrow_HIGrow/fcf5c40cfb10 b'{"plant":{"Tgrow_HIGrow":"fcf5c40cfb10","sensorname":"driveway_lawn","date":"30-09","time":"10:03","sleep5Count":0,"bootCount":0,"lux":523.3333,"temp":27.8,"humid":59,"soil":1,"soilTemp":0,"salt":0,"saltadvice":"needed","bat":100,"batcharge":"charging","batchargeDate":"30-09","daysOnBattery":0,"battvolt":2768,"battvoltage":4.90737,"pressure":0,"plantValveNo":1,"wifissid":"-REDACTED-","rel":"4.3.2"}}'
in read_yaml
Traceback (most recent call last):
  File "TTGO-T-HiGrow-aut.py", line 210, in <module>
    client.loop_forever()
  File "/usr/local/lib/python3.6/site-packages/paho/mqtt/client.py", line 1779, in loop_forever
    rc = self.loop(timeout, max_packets)
  File "/usr/local/lib/python3.6/site-packages/paho/mqtt/client.py", line 1181, in loop
    rc = self.loop_read(max_packets)
  File "/usr/local/lib/python3.6/site-packages/paho/mqtt/client.py", line 1572, in loop_read
    rc = self._packet_read()
  File "/usr/local/lib/python3.6/site-packages/paho/mqtt/client.py", line 2310, in _packet_read
    rc = self._packet_handle()
  File "/usr/local/lib/python3.6/site-packages/paho/mqtt/client.py", line 2936, in _packet_handle
    return self._handle_publish()
  File "/usr/local/lib/python3.6/site-packages/paho/mqtt/client.py", line 3216, in _handle_publish
    self._handle_on_message(message)
  File "/usr/local/lib/python3.6/site-packages/paho/mqtt/client.py", line 3444, in _handle_on_message
    self.on_message(self, self._userdata, message)
  File "TTGO-T-HiGrow-aut.py", line 196, in on_message
    yaml_data = read_yaml_file(mac_id, name)
  File "TTGO-T-HiGrow-aut.py", line 36, in read_yaml_file
    line_mac_id = line_mac_id_arr[1]
IndexError: list index out of range
`

riddlecp, 2021-09-30:
This doesn't really make any sense to me, but it worked - change the line 
`line_mac_id = line_mac_id_arr[1]`
to
`line_mac_id = line_mac_id_arr[0]`

pesor, 2021-09-30:
I look into it, how does your sensors.yaml look like. 

//Per

riddlecp, 2021-09-30:
after changing that index value, it created fine apparently:

`# cat sensors.yaml`
`info: This file contains the MAC_ID of the sensors already registred for Autodiscover`
`info: You can reinitiate Autodiscover for a single MAC_ID, by removing it from the list, or you can initiate a full renewal of sensors to Autodiscover,`
`info: By removing all MAC_IDs below these comments lines`
`info: DO NOT REMOVE THESE FOUR INFO LINES.`

`mac_id: fcf5c40cfb10`
`name: driveway_lawn`