pesor/TTGO-T-HIGrow #29: Fertilizer reporting in percentage rather than µS/cm

netmanchris, 2022-03-07:
First - Thanks for this great project. Really appreciate you and @BeardedTinker working on this! 

I’m noticing that the output from the HiGrow sensor for the fertilization level is retuning a % value. But strangely - it looks like the value can sometimes be high ( seeing 399% on one sensor. )  Given that I’ve mapped the sensor._name_fertilizer into the conductivity key for the HomeAssistant Plant integration. Should we think of the fertilizer value as mapping direction to the µS/cm value? Ie. The max/min conductivity values for the Plant integration should still apply?

pesor, 2022-03-08:
I more or less gave up on this value, as it is not being estimater correctly, because it is using DC in stead of AC. 

This means that the value decreases rapidly and not according to a real measurement. 

I Seem to remember that I have explained it in the WIKI.

kohlerryan, 2023-07-18:
Interested in this issue as well.  Would be nice if it could work.