pesor/TTGO-T-HIGrow #30: Not using with Home Assistant

Ianszh, 2022-04-10:
Hello, I'd like to use this outside of Home Assistant (MQTT - Node-Red), I have some of the older versions with a 18650 battery and  DHT11's, I've set the necessary true and false in user-variables.h and  I'm receiving the MQTT messages in node-red, but there are some missing and strange values.
No data from the DHT11 and daysOnBattery: 19092.7 etc. could you give some advice please?

`"{"plant":{"Tgrow_HIGrow":"246f28af39e8","sensorname":"Chilli-1","date":"10-04","time":"17:31","sleep5Count":0,"bootCount":34,"lux":-2,"temp":null,"humid":null,"soil":65535,"soilTemp":0,"salt":0,"saltadvice":"needed","bat":-220,"batcharge":"","batchargeDate":"","daysOnBattery":19092.7,"battvolt":69,"battvoltage":0.12233,"pressure":0,"plantValveNo":1,"wifissid":"xxxxx","rel":"4.3.2"}}"`

pesor, 2022-04-10:
In the uservariable file there are a lot of variables, have you looked into this.

One set concerns the DHT11.

//
Per

Sendt fra min iPhone

Ianszh, 2022-04-11:
Yes, I did look into the variables when setting up. any Idea about the daysOnBattery?

Also is there a setting where I can adjust the wake time?

pesor, 2022-04-19:
You can only assign the sleep time, see the WIKI to learn more.