pesor/TTGO-T-HIGrow #31: WiFi connected but no DHCP

TheCranston, 2022-04-26:
I'm not sure what happened other than I made a change to the sleep length and re-uploaded...

I see the esp32 connect to my WiFi access point, however I don't see a DHCP request going out.
The AP shows an association but no IP address.
The esp32 is "stuck" with this continuous error in the serial console:
```[xxxx][E][WiFiUdp.cpp:170] beginPacket(): could not get host from dns: 11```
where the [xxxx] is an ever increasing number.  

I've tried using esptool.py to erase the SPI flash and start over thinking something got corrupted there.  No change in behaviour.
I'm thinking something is out of compatibility spec in the platform.io environment, but alas I'm lost there.  really just following BeardedTinker's video/docs on how to build the project.  It did auto update a bunch of libraries and maybe that's the issue.  Any help is apprecaited.

I've also tried fully ripping out VSC and platform.io and starting over.  Same result.  Not sure what is breaking.
pio core 5.2.5 home 3.4.1 with espressif 4.1.0  

I know the WiFi is physically working as I went back to Arduino IDE and was able to get the http client example sketch to work just fine.

And some pause before submitting and a little google searching and I find a recommended adjustment:
```
#include <WiFi.h>
void setup() {
  Serial.begin(115200);
  WiFi.begin(); //Put ssid, passwd in here if needed
  WiFi.waitForConnectResult();
  Serial.println(WiFi.dnsIP());
  WiFi.config(WiFi.localIP(), WiFi.gatewayIP(), WiFi.subnetMask(), WiFi.dnsIP()); 
  delay(10);
  Serial.println(WiFi.dnsIP());
}
```
Tested locally and can confirm it works now.

townet01, 2022-05-03:
Is necessary to comment this:

 //WiFi.config(INADDR_NONE, INADDR_NONE, INADDR_NONE); // call is only a workaround for bug in WiFi class

TheCranston, 2022-05-03:
So @townet01 you are saying that commenting out the line has the same effect as what I found of changing
```
WiFi.config(INADDR_NONE, INADDR_NONE, INADDR_NONE);
```
to
```
WiFi.config(WiFi.localIP(), WiFi.gatewayIP(), WiFi.subnetMask(), WiFi.dnsIP()); 
```

Works just as well as
```
//WiFi.config(WiFi.localIP(), WiFi.gatewayIP(), WiFi.subnetMask(), WiFi.dnsIP()); 
```

correct?

townet01, 2022-05-03:
Yes is correct !
Try.

townet01, 2022-05-03:
Yes it's correct !

E.

TheCranston, 2022-05-03:
I can try it this upcoming weekend.  I’ll report results then.  Thanks

JonOve, 2022-05-16:
Connecting the board to wifi I see the router gives the subnet mask adress 255.255 etc. as an ipadress to the board.
Adding above mentioned line to "user-variables.h" does not solve the issue.
Any ideas?

TheCranston, 2022-05-16:
@JonOve can you provide logs of the DHCP request from your router?  The router *should* not be handing out 255.255.255.255 as an IP address or something else is very wrong.  The suggested edit is in the "connect-to-network.h" file right around line 7 in the "connectToNetwork()" function.

JonOve, 2022-05-16:
@TheCranston, thank you for your quick reply. I was not aware of a "connect-to-network.h" file. Will see if I can find that one and edit it as proposed.

JonOve, 2022-05-16:
@TheCranston, after the edit the board connected to the ssid properly with a correct ipadress. Many thanks for your support!