pesor/TTGO-T-HIGrow #32: not an issue, mqtt autodiscovery idea for ha

Maaciej, 2022-05-12:
Auto discovery is complicated, it is not working for me, I don't understand what it tries to do, so I played with code.
***************
save-configuration.h

new function
***********************************************
```
void mqttSetup(String identyfikator, String chipId, String uom = "x" )
{

  Serial.println("*****************************************" );

  const String topicStr_c = "homeassistant/sensor/" + device_name + "-" + chipId + "/" + identyfikator +"/config";
  const char* topic_c = topicStr_c.c_str();
  Serial.println(topic_c);

  StaticJsonDocument<1536> doc_c;
  JsonObject root = doc_c.to<JsonObject>();

  root["name"] = device_name +" "+ identyfikator;
  root["state_topic"] = "homeassistant/sensor/" + device_name + "-" + chipId  + "/status";
  root["value_template"] = "{{ value_json['" + identyfikator +"'] }}";
  if ( uom != "x" ) {
    root["unit_of_measurement"] = uom;
  }
//copy of existing sending code
// // Send to mqtt
  char buffer_c[1536];
  serializeJson(doc_c, buffer_c);

  Serial.print("Sending message to topic: ");
  if (logging) {
    writeFile(SPIFFS, "/error.log", "Sending message to topic: \n");
  }

  Serial.println(buffer_c);

  // Connect to mqtt broker
  Serial.print("Attempting to connect to the MQTT broker: ");
  if (logging) {
    writeFile(SPIFFS, "/error.log", "Attempting to connect to the MQTT broker! \n");
  }

  // Serial.println(broker);
  mqttCli

ent.setServer(broker, port);

  if (!mqttClient.connect(broker, mqttuser, mqttpass)) {
    if (logging) {
      writeFile(SPIFFS, "/error.log", "MQTT connection failed! \n");
    }

    Serial.print("MQTT connection failed! Error code = ");
    Serial.println(mqttClient.state());
    // goToDeepSleepFiveMinutes();
  }

  if (logging) {
    writeFile(SPIFFS, "/error.log", "You're connected to the MQTT broker! \n");
  }

  Serial.println("You're connected to the MQTT broker!");
  Serial.println();

  bool retained = true;

  if (mqttClient.publish(topic_c, buffer_c, retained)) {
    Serial.println("Message published successfully");
  } else {
    Serial.println("Error in Message, not published");
    // goToDeepSleepFiveMinutes();
  }
  Serial.println("*****************************************" );  
  Serial.println();

}
```
*******************************************
called before making json for mqtt with measure values:
*********************************************
```
  Serial.println("chipId " + chipId); //existing line

  mqttSetup("daysOnBattery", chipId);
  mqttSetup("battvoltage", chipId, "V");
  mqttSetup("bat", chipId, "%");
  mqttSetup("lux", chipId, "lx");
  mqttSetup("humid", chipId, "%");
  mqttSetup("soil", chipId, "%");
  mqttSetup("salt", chipId);
  mqttSetup("temp", chipId, "°C");

const String topicStr = "homeassistant/sensor/" + device_name + "-" + chipId + "/status"; //existing line
```

And with this changes measurements show itself in home assistant for me:

[image]

still don't know how to make the measurements  to be possible to disable, I mean make them with unique entity_id

[image]

***************
And question, 
it seems like there somewhere is log stored and other files, how to access them? 
I thought there is web server, or it should be visible as a portable drive but I can't find.
Can anybody explain how to access log?

Maaciej, 2022-05-12:
one more thing, because it can matter
I changed original JSON to this, with one level structure:

```
StaticJsonDocument<1536> doc;
  // Set the values in the document
  // Device changes according to device placement
  JsonObject plant = doc.to<JsonObject>();
  
  
  // JsonObject plant = root.createNestedObject("sensor"); //CHANGE!!!!!
  plant[device_name] = chipId;
  plant["sensorname"] = plant_name;
  plant["date"] = config.date;
  plant["time"] = config.time;
  plant["batchargeDate"] = config.batchargeDate;
  plant["daysOnBattery"] = config.daysOnBattery;
  // plant["battvolt"] = config.batvolt; //nie
  plant["bat"] = config.bat;
  plant["battvoltage"] = config.batvoltage;
  plant["sleep5Count"] = config.sleep5no;
  plant["bootCount"] = config.bootno;
  // plant["batcharge"] = config.batcharge; //nie
  plant["lux"] = config.lux; //nie
  plant["temp"] = config.temp;
  plant["humid"] = config.humid;
  // plant["pressure"] = config.pressure;
  plant["soil"] = config.soil;
  // plant["soilTemp"] = config.soilTemp; //nie
  plant["salt"] = config.salt;
  // plant["saltadvice"] = config.saltadvice;//nie
  // plant["plantValveNo"] = plantValveNo; //nie
  // plant["wifissid"] = WiFi.SSID(); //nie
  plant["rel"] = config.rel;
```

with two level structure one have to change json templates to read measurements.

Maaciej, 2022-05-13:
entity_id solved.
Because there are lot of changes in the file, full content:

```
void mqttSetup(String identyfikator, String chipId, String uom = "x", String dc = "x" )
{

  Serial.println("*****************************************" );

  const String topicStr_c = "homeassistant/sensor/" + device_name + "-" + chipId + "/" + identyfikator +"/config";
  const char* topic_c = topicStr_c.c_str();
  Serial.println(topic_c);

  StaticJsonDocument<1536> doc_c;
  JsonObject root = doc_c.to<JsonObject>();

  root["name"] = device_name +" "+ identyfikator;

  if ( dc != "x" ) {
    root["device_class"] = dc;
  }

  root["unique_id"] = chipId +"-"+ identyfikator;
  root["state_topic"] = "homeassistant/sensor/" + device_name + "-" + chipId  + "/status";
  root["value_template"] = "{{ value_json['" + identyfikator +"'] }}";
  if ( uom != "x" ) {
    root["unit_of_measurement"] = uom;
  }

// // Send to mqtt
  char buffer_c[1536];
  serializeJson(doc_c, buffer_c);

  Serial.print("Sending message to topic: \n");
  if (logging) {
    writeFile(SPIFFS, "/error.log", "Sending message to topic: \n");
  }

  // Serial.println(buffer_c);
  serializeJsonPretty(doc_c, Serial);
  Serial.println();

  // Connect to mqtt broker
  Serial.print("Attempting to connect to the MQTT broker: ");
  if (logging) {
    writeFile(SPIFFS, "/error.log", "Attempting to connect to the MQTT broker! \n");
  }

  // Serial.println(broker);
  mqttClient.setServer(broker, port);

  if (!mqttClient.connect(broker, mqttuser, mqttpass)) {
    if (logging) {
      writeFile(SPIFFS, "/error.log", "MQTT connection failed! \n");
    }

    Serial.print("MQTT connection failed! Error code = ");
    Serial.println(mqttClient.state());
    // goToDeepSleepFiveMinutes();
  }

  if (logging) {
    writeFile(SPIFFS, "/error.log", "You're connected to the MQTT broker! \n");
  }

  Serial.println("You're connected to the MQTT broker!");
  Serial.println();

  bool retained = true;

[... 110 lines cut ...]
  }

  if (logging) {
    writeFile(SPIFFS, "/error.log", "You're connected to the MQTT broker! \n");
  }

  Serial.println("You're connected to the MQTT broker!");
  Serial.println();

  bool retained = true;

  if (mqttClient.publish(topic, buffer, retained)) {
    Serial.println("Message published successfully");
  } else {
    Serial.println("Error in Message, not published");
    goToDeepSleepFiveMinutes();
  }
  Serial.println();
}

```

and nice result in entities

[image]

Maaciej, 2022-05-17:
> Is it the intention to replace the content of the file "save-configuration.h" with your code as mentioned above?

yes this whole save-configuration file content customized (unnecessary fields of json commented out and changed json structure) for my needs, with added mqtt autodiscovery for ha.
I guess you change your file to above and should see that ha will detect configured metrics.

JonOve, 2022-05-18:
Indeed, worked like a charm. Thanks a lot for the effort!

pesor, 2022-05-21:
I replaced the save-configuration.h with your content above, but I get this error, any advice:

Creating JSON document...
chipId 246f28b21b3c
*****************************************
homeassistant/sensor/Tgrow_HIGrow-246f28b21b3c/daysOnBattery/config
Sending message to topic:
{
  "name": "Tgrow_HIGrow daysOnBattery",
  "device_class": "duration",
  "unique_id": "246f28b21b3c-daysOnBattery",
  "state_topic": "homeassistant/sensor/Tgrow_HIGrow-246f28b21b3c/status",
  "value_template": "{{ value_json['daysOnBattery'] }}"
}
Attempting to connect to the MQTT broker: Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (loopTask)
Core 1 register dump:
PC      : 0x40093994  PS      : 0x00060036  A0      : 0x4008e638  A1      : 0x3ffc8e20
A2      : 0x00000001  A3      : 0x3ffc8050  A4      : 0x3ffc8050  A5      : 0x00000001
A6      : 0x00060020  A7      : 0x00000000  A8      : 0x80086148  A9      : 0x3ffc8ec0
A10     : 0x3ff000e0  A11     : 0x00000001  A12     : 0x3ffbe9b8  A13     : 0x00000001  
A14     : 0x00060023  A15     : 0x00000000  SAR     : 0x00000018  EXCCAUSE: 0x00000001
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

ELF file SHA256: 0000000000000000

Backtrace: 0x40093994:0x3ffc8e20 0x4008e635:0x3ffc8f00 0x40127f71:0x3ffc8f40 0x40128a8f:0x3ffc8f60 0x401282b9:0x3ffc8f90 0x4012846c:0x3ffc8fb0 0x4011f158:0x3ffc8ff0 0x400d4cc4:0x3ffc9010 0x400d45c6:0x3ffc9070 0x40164a79:0x3ffc90a0 0x400d835a:0x3ffc90c0 0x400d8542:0x3ffc9100 0x400d2401:0x3ffc9130 0x400d2765:0x3ffc9dd0 0x400d4167:0x3ffcac50 0x400db5ea:0x3ffcad80 0x40091282:0x3ffcada0

Rebooting...
ets Jun  8 2016 00:22:57

Maaciej, 2022-06-23:
> And question,
it seems like there somewhere is log stored and other files, how to access them?
I thought there is web server, or it should be visible as a portable drive but I can't find.
Can anybody explain how to access log?

files in question are:

```
Listing directory: /
  FILE: favicon.ico     SIZE: 13532
  FILE: index.html      SIZE: 482
  FILE: app.js.gz       SIZE: 198509
  FILE: batinfo.conf    SIZE: 21
  FILE: error.log       SIZE: 1342
  FILE: soil.conf       SIZE: 9
  FILE: name.conf       SIZE: 9
```
files with extension .conf are from the program, and unknown are the first three.
I suspect now they are from the original firmware.
They cannot be removed by "remove",
but I erased all files with `SPIFFS.format()`
and now have only new files:

```
  FILE: name.conf       SIZE: 9
  FILE: batinfo.conf    SIZE: 21
  FILE: soil.conf       SIZE: 9  
```

elgansayer, 2023-02-18:
Hi, i found a few issues with the above. Mainly having a device and sensor names. So updated some, added some,.

[image]

https://github.com/elgansayer/TTGO-T-HIGrow/tree/feature/AutoDiscovery